frc-862/valkyrie #10: Ramp power to motors

phurley, 2017-01-11:
With a 6CIM drive train, starting from a stop, the motors will draw an enormous amount of current if we command 0% to 100%. On a full battery this might work, but by the end of match, it would likely cause the robot to brown out (which we may see as a "shutter" as the system attempts to save itself, but killing all motor power, which causes voltage to climb, then re-enabling motors, which causes it to fall again, and repeating many times a second).

The solution to this is ramp power to the motors. We will need to calibrate the rate, but the idea is simple and we can have the talons do all the real work for us.

pdbingham, 2017-01-11:
Additionally, with 6 CIMs I would expect a significant hit on current draw
in low gear when we are trying to push another robot.

I understand wanting more power and speed, but we need to monitor the
battery life closely.
I'm not convinced we "need" 6 CIMs.  We had a lot of power and speed with
Orion.  We should have the option (in design) to scale it back to a 4 CIM
design if necessary.

phurley, 2017-01-11:
We definitely do not need 6 CIMs, but they will provide the following:

- Better acceleration (more torque)
- More resistance in the drive train: when we are in low gear and not moving or moving slowly with the motors in brake mode it will take a not insignificant impact to move us, which will help when trying to place gears

In low gear we should be *traction limited* that is to say, above a certain power input our wheels will being to slip -- we won't be winning the pushing battle at that point, but we won't stall/brownout either. This is by design -- and necessary even with a 4 CIM solution.

phurley, 2017-01-20:
Ramping implementing using talons in meeting

closed 2017-01-20 by phurley